omkbd/ErgoDash #16: Facing issues sourcing the MJ-4PP-9

Strayer, 2019-05-22:
Hey, this is a little off topic, sorry for that. I'm having issues sourcing the MJ-4PP-9 TRRS jack in Europe (or anywhere outsite of Japan, to be precise).

I ordered some 4 pol TRRS jacks, but the connections on those will require some hacking on the PCB – so any tips where I could find the MJ-4PP-9 internationally would be greatly appreciated!

The ones I ordered are these:
FC68125
https://cdn-reichelt.de/documents/datenblatt/C160/4%20POLE%20JACKS.pdf

GreenShadowMaker, 2019-05-23:
20PCS DIP PJ320A 3.5MM Headphone Audio Socket PJ-320A PJ320 PJ-320 4 pins Audio interface
https://s.click.aliexpress.com/e/pGyft1Q

Is what I used. (Fits perfect) After getting a url from omkbd

Strayer, 2019-05-23:
Hey @GreenShadowMaker, thanks for confirming! Those are the ones I purchased on AliExpress, so I guess I'm fine. I'm still debating wether I should modify the footprint, but the sizes are off by such a tiny amount (0.1mm - 0.2mm) that I might just risk it.

closed 2019-06-05 by omkbd