omkbd/ErgoDash #17: Piezo speaker placement

brethash, 2019-05-23:
I got in on the most recent [group buy](https://evolve-keyboards.myshopify.com/products/ergodash-mechanical-keyboard-kit-pre-order) and grabbed a piezo speaker to add to my build (since the group buy page listed it as an option) but I don't see it mentioned anywhere in the [build guide](https://github.com/omkbd/ErgoDash/blob/master/Doc/build-en.md). I did look back through the past issues for this repo and noticed issue #7. The only comment in there was that the speaker isn't support while using the RGB LEDs. Is this still true?

Thanks for the help as well as maintaining this project!

omkbd, 2019-05-24:
I have no plans to support speakers in the future. If you use both RGB LEDs, you may be able to use it by reducing the number of RGB animation patterns.

brethash, 2019-05-24:
Ah ok. I guess I have an extra piezo speaker on my hands!

closed 2019-05-24 by brethash