omkbd/ErgoDash #18: LEDs and inconsistant keys

nutimnot, 2019-05-23:
Hello,

I built this board yesterday and have been having some issues. First I cannot figure out how to turn the LEDs on. I flashed my custom .hex file to the keyboard but other than that i cannot get the lights to turn on. I update the file like the instructions say but i'm not sure how they're supposed to populate to the board itself.

Secondly i've been having some issues getting keys to work sometimes. For example when i built the board, I couldn't get my 7 key to work, i thought I had soldered it incorrectly but everything else works. Now the 7 works fine and the enter key doesn't work without changing anything. Also sometimes the keys need to be pressed a few times before they'll register with the computer. Any assistance would be greatly appriciated, thank you!

omkbd, 2019-05-24:
・First question
Press BL_TOGG or RGB_TOG.

・Second question
I can only think of checking the soldering.

nutimnot, 2019-05-24:
Thanks for the reply! How do I press BL_TOGG or RGB_TOG?

brethash, 2019-05-24:
@nutimnot could you post a couple pictures of the boards? It might help rule out soldering as the culprit.

nutimnot, 2019-05-24:
@brethash here are some pictures. I got home just now and added some more solder to the mis-behaving keys and reflashed that side of the keyboard to no avail, the entire HJKL:" row is dead.

for reference, in the picture the middle row is the HJKL:" row
[image]
[image]

GreenShadowMaker, 2019-05-25:
I haven't had working RGB LEDs on either of the 2 I built.. I'm pretty sure I screwed up my first build.(backwards on one side) not sure why my second didn't work.

But, in your photos you haven't done the RGB jumper connetion.

GreenShadowMaker, 2019-05-25:
For an entire row that stopped working, it may be the pro-micro connection.

nutimnot, 2019-05-25:
@GreenShadowMaker how would i go about fixing that kind of issue?

nutimnot, 2019-05-30:
I managed to fix the issues with the HJKL row. One of the points on the Pro micro wasn't soldered well, so I remelted it and now it's fully functional (typed this reply on the keyboard). If anyone knows why the LEDs aren't working let me know. I don't have any resistors set up, the kit only came with diodes.

brethash, 2019-06-03:
@nutimnot you won't need any additional resistors for the undermount LEDs to work. The resistors are only needed for the backlight/switch LEDs. 

If you're using the default keymap for the keyboard, try flashing @GreenShadowMaker 's keymap. It seems like his has the undermount LEDs on by default (at least thats what it seemed like when I flashed it anyway). The layers in his keymap are also a bit easier to work with in my opinion.

nutimnot, 2019-06-03:
@brethash where can I find their keymap?

brethash, 2019-06-03:
The keymap is [here](https://github.com/qmk/qmk_firmware/tree/master/keyboards/ergodash/rev1/keymaps/greenshadowmaker) and the layout itself is [here](https://github.com/qmk/qmk_firmware/blob/master/keyboards/ergodash/rev1/keymaps/greenshadowmaker/keymap.c). To toggle the `adjust` layer, the raise and lower keys are pressed at the same time (for me they were the keys directly below the `B` and the `N`.

nutimnot, 2019-06-03:
Awesome thank you! Now I'd just try flashing it with QMK and see if it works yeah?

brethash, 2019-06-03:
Yep! You should be able to build/flash it with `make ergodash/rev1:greenshadowmaker:avrdude` from the root of the `QMK` folder.

nutimnot, 2019-06-05:
@brethash how do you do that using the QMK toolbox?

brethash, 2019-06-05:
@nutimnot that's a good question! I've actually not used the QMK toolbox before; I always just build and flash from the command line, so I'm not sure how it works. 😞

nutimnot, 2019-06-05:
I flashed it but it didn't turn on the LEDs so i'm not sure. Maybe I will try something else or test to see if the LEDs work at all I guess. Thank you @brethash for all your help my guy

omkbd, 2019-06-05:
You haven't done the RGB jumper connetion.
You have to read this.
https://github.com/omkbd/ErgoDash/blob/master/Doc/build-en.md#5-underglow-led-optional

nutimnot, 2019-06-05:
@omkbd So I just have to solder those two connections at the top of the board? I shouldn't need anything other than some solder, right? I don't have any resistors installed or anything. I'll try it out this week

nutimnot, 2019-06-07:
I got it to work, thank you @omkbd and @brethash for y'alls help <3

closed 2019-06-07 by nutimnot