omkbd/ErgoDash #19: No right side backlight LEDs lit

brethash, 2019-05-24:
I was just finishing up my Ergodash and wanted to test the backlight LEDs before I went ahead and soldered all 34 onto the boards. I noticed that the right side LEDs don't seem to light up, but the left side (my master side in this case) appear to all work as expected. I loaded up the `greenshadowmaker` keymap, but I don't really see any sort of configuration that would indicate that _only_ the left side should be backlit. 

Before I go and desolder the mosfet, I wanted to double check that there isn't any firmware options I should check. Additionally, I'm 99% sure that the polarity is correct on the 3 LEDs I installed on the right. I even tried soldering one in with the reverse polarity and one with the correct polarity just to prove to myself that I've got them in right!

Thanks again for any help!

GreenShadowMaker, 2019-05-25:
The mosfet is missing in this photo isn't it?

brethash, 2019-05-28:
Hey @GreenShadowMaker ! First of all, thanks for putting together this keymap! 

The mosfet is on the board, just above the rightmost RGB LED on bottom row of the RGB LEDs. For what its worth, I did add just a touch more solder to the connection on the side with one lead (the right side of mosfet in this image), but I got the same result.

GreenShadowMaker, 2019-05-29:
Ah, it just blended in too well in the photo. I can see solder.. but the black chip outline was hard :)

I built 1.1 that doesn’t have the extra resistor next to the bl one. Looks like it may be for the speaker?

brethash, 2019-06-03:
So I got back home after the holiday weekend and removed the old mosfet and tried another in its place. Same result. :( 

I also double checked all the resistors on the board and they all appear to be in the correct location according to the left side (that works!) and the updated build guide. I suppose I'll just solder on the rest of the backlight LEDs for fun and hope for the best.

omkbd, 2019-06-13:
I think that it is a solder defect of MOSFET.