omkbd/ErgoDash #22: Idea: Rotary Encoder

dmcgits, 2019-06-22:
Have you ever considered one of these for the ergodash? Apologies that I’m suggesting rather than attempting it myself, I’m still new to circuits/kiCad. 

The Preonic’s rotary encoder is pretty neat and I’d love to replace maybe my top row innermost keys (currently -, + in the laser image) with a rotary encoder. Alternatively, it could be thumb accessible on the snap-off section.

The Preonic uses a twisting stick style dial, but with an ortho being so friendly to home row users it’d be slick if it were a drum you could roll with a thumb or forefinger without flexion.  Like a Logitech G810 but the width of a keycap :)

Since Jack added one to Preonic I’m guessing QMK support is there too.

Thanks for reading and thanks again for your generosity!

brethash, 2019-06-23:
For what its worth, I'd love something like this and have been thinking about if it could work. I was literally just telling a friend of mine today how much I've been loving my Ergodash, but that I was considering building an Iris just to play around w/ the addition of a rotary encoder. 

I did notice during the build that there seem to be some indications on the PCB for an I2C interface, but I'm not 1000% sure how to access that. If that were possible, you could theoretically leverage it to add an I2C-ready rotary encoder like [this](https://hackaday.com/2018/04/15/rotary-encoders-become-i2c-devices/), which I think you can buy [here](https://www.ebay.com/itm/I2C-Encoder-V2-RGB-Encoder-Knob-connector/173696523407?_trkparms=aid%3D888008%26algo%3DDISC.CARDS%26ao%3D1%26asc%3D20131227121020%26meid%3D4a3249bca6194a5faec3bdf56a84e0dd%26pid%3D100009%26rk%3D1%26rkt%3D1%26sd%3D333148825152%26itm%3D173696523407&_trksid=p2047675.c100009.m1982) (I think!). 
@omkbd I don't suppose you might be able to shed a little light on this potential implementation?

omkbd, 2019-06-23:
I am already working on the implementation of the rotary encoder. However, I am worried about where to place it. I think that the position of the thumb is the best.

brethash, 2019-06-23:
I was also thinking somewhere around the thumb cluster. I'm curious how you were going about the implementation? Would this be an add-on for the current v1.2 board or would this be something for a v1.3? I was thinking of just whipping up a little enclosure for the encoder and its pcb that attaches with the top case plate via the screw at the top of the thumb cluster. Then I was just going to throw it on my 3d printer and see what it takes me. :)

dmcgits, 2019-06-24:
Thanks so much @omkbd for responding, also @brethash!

What if you had a V1.2 and a V1.2R, the R being a version with rotary encoder?

I’m having a think about the positioning and will sketch them down, maybe it’ll be useful.

omkbd, 2019-07-04:
Testing...
[image]

dmcgits, 2019-07-04:
Haha, lovely!

dmcgits, 2019-10-06:
Hi @omkbd, how did the testing go? Is there somewhere you’d rather I ask questions? Maybe Discord?

MattBlissett, 2019-11-12:
In case anyone else wants to hack this on an existing board:

[image]

[image]

I'm not using the backlight or audio, so these two pins are available, and there's conveniently a trace bringing them across the board.

It's then simple to add support in QMK: https://github.com/MattBlissett/qmk_firmware/commit/16ed43b4dc5c786f69ceaa68fbfaec1b29208507

The encoders are jammed into the MX plate holes with some bits of folded cardboard. Since the keyboard is going to stay on my desk, this is probably OK for a while.

> whipping up a little enclosure for the encoder and its pcb that attaches with the top case plate via the screw at the top of the thumb cluster

That would be much neater. Another option is to 3D print or use a tiny PCB that fits in the MX hole, like [this one](https://www.reddit.com/r/MechanicalKeyboards/comments/chs82g/designed_a_new_adapter_for_installing_rotary/) (except that's for handwired keyboards).

> Will it be difficult to make it work on both sides of pcb like your switch mounts do?

I doubt it, as the wiring is symmetrical:

```
A C B 
| | | 
[===]
 | |
 D E
```  

`C` is the common pin, `A` and `B` are the signals when rotating. `E` and `F` are the push button.

n8gray, 2019-12-04:
Oh dear lord, now I can't live without this.  I've been debating between Redox and ErgoDash but this decides it.

dyamon, 2021-02-09:
Hey @MattBlissett , noob-ish guy here! Would you mind giving a bit more details on how to connect the wires on the encoder side? Do you get both rotation and push botton functionalities?

MattBlissett, 2021-02-09:
Hi,

I don't use the push button of the encoder -- the ones I bought are rather stiff to push, so I didn't feel I needed it.  There's also nowhere to connect it, unless you swapped one of the nearby keys for a double-height one and used the freed-up connection, but I don't see the benefit.

There was a typo in my comment above, which I've fixed (above too):

```
A C B 
| | | 
[===]
 | |
 D E
```  

The middle pin `C` is the common pin, which I connected to the unused audio trace.  `A` and `B` are the signals when rotating, they are activated in a special pattern ([Gray code](https://en.wikipedia.org/wiki/Gray_code)) when rotating the knob.  These I connect to the backlight trace:

[image]

If you connect `A` and `B` the "wrong" way around, just change the key mappings in your QMK code accordingly.

dyamon, 2021-02-09:
Thank you very much!

> There's also nowhere to connect it, unless you swapped one of the nearby keys for a double-height one and used the freed-up connection, but I don't see the benefit.

Oh I see, that's because you installed it in the extra thumb key, which I removed. I'm planning on putting it next to it where I have the connection directly under the encoder. So it seems like I might be able to get the push button functionality as well.

> The middle pin `C` is the common pin, which I connected to the unused audio trace. `A` and `B` are the signals when rotating, they are activated in a special pattern ([Gray code](https://en.wikipedia.org/wiki/Gray_code)) when rotating the knob. These I connect to the backlight trace

Thanks, this is super useful!

gregor-on, 2023-02-22:
it can be done relatively easy, by utilising led's pins from the board Arduino, and also I've been able to add displays by changing D0 to D2 and attaching display there see at gdoes.art , I'm not great at this github thing, not sure how to share it. message me and I'd like to attach it here including CAD files etc. btw I used vertical encoders, mouse - like setup. Ergodash is a great platform to play with, it is a shame it is so hard to get it working with VIA tho