omkbd/ErgoDash #24: Right half keys not working

callebjorkell, 2019-07-29:
I've been struggling to get the right half of my ergodash to work, and at this point I cannot figure out if it's a hardware or a software issue. When I plug the keyboard in, the LEDs turn on, and the keys on the left side works, while the keys on the right hand side does nothing.

My situation is:

* LEDs on both sides turn on when I plug in the TRRS cable
* I'm using sparkfun pro micros
* TRRS pins have continuity (checked with a multimeter) between all four pins. Maybe someone could tell me which pins on the pro micro they are connected to so I can check all the way from pro micro to pro micro? Can't really follow the traces.
* Both halfs work if I plug them in directly with the USB cable
* I've tried various settings for qmk to try to troubleshoot, but nothing has made a difference, and right now the only tweak I have is `RGBLIGHT_ENABLE = yes` in `rules.mk`
* building and flashing with `sudo make ergodash/rev1:default:avrdude` on latest qmk master (eafd38e2a16cd00e95e00c5ca4efc364d3ccf3e4)

Are you able to point me towards what I could do to try to troubleshoot, and/or confirm that I've got everything soldered in the right place?

Front:
[image]

Back:
[image]

Appreciate the help/time!

callebjorkell, 2019-07-29:
I think I found the pin as well. I have a good connection between the pin marked "SCL - 3" in this diagram: [image]

I hope that sounds right?

GreenShadowMaker, 2019-08-01:
https://imgur.com/a/ect63fa
Is the back of mine, working.

Pro micro orientation would seem to be fine

-I don’t think I tried mine before I had all the switches installed

Probably not it since your rgb light up.(mine didn’t from what I remember. but remove the solder from the selector pads. For the rgb on the master and see if it works.   I think I overheated/fried my rgb or reversed one, I didn’t get them to work. But my second half had issues until removed jumper solder from master.

callebjorkell, 2019-08-01:
Tried to desolder the pads, but unfortunately there is no change to how the keys work, but LEDs are now off of course.

callebjorkell, 2019-08-07:
It seems likely that this issue is caused by my use of the sparkfun pro micro. I tried to mount a "Tenstar robot" pro micro onto the half that wasn't working before, and now it works fine as slave, but the one with the sparkfun pro micro still does not work as a slave, even though it's fine as master. I even tried to build a spare PCB that I had, put the sparkfun pro micro on that, and it exhibits the exact same problems as the original one I built. In short, I have three halves:
* 1 and 2 with sparkfun => both work fine as master, neither as slave
* 1 with tenstar, 2 with sparkfun => 1 works fine as either slave or master, 2 only as master.
* 3 and 2 with sparkfun => both work fine as master, neither as slave. Both work fine as master and the tenstar 1 as slave.

I chose the sparkfun on color alone, since I have an acrylic case and wanted the white PCB with the red pro micro, so this feels silly now. Does anyone know in which way the sparkfun pro micro is different, and possibly a way to work around this issue to make both my spark fun pro micro halves work? I'm hoping this can be corrected in software. I've got the controllers soldered onto the board, so I doubt I'll be able to get them desoldered without maiming the board. :disappointed:

callebjorkell, 2019-09-09:
Gave up, made another one with an elite-c, works fine now. I'll close the issue as I scavenged the these boards for parts, and hence can't confirm that any suggested fix is working anymore anyway. Still irks me that I don't know what was wrong with it though.

closed 2019-09-09 by callebjorkell

Strayer, 2020-01-20:
So, I'm having the same problem with my build. It found references that Sparkfun Pri Micros have J1 jumpers jumped and need to be desoldered for Lets Split keyboards to work, maybe this is the same issue here? I'm probably not using genuine Sparkfun Pro Micros, but still will have to check that :/

nchrisdk, 2020-02-15:
I had the same issue. I just "desoldered" one of my failed attempts and removed (with force) Pro Micro. J1 jumper was not desoldered...

[image]

JonahBranch, 2022-09-27:
I'm having problems with the right half as well. Made 3 sets of keyboards and only the left hand will flash. The right hand always responds with "programmer not responding" when trying to flash, and that's when it's plugged in directly to the computer.

It's possible that those microcontrollers happen to be defect, but the fact it happened 3 times on the right hand makes me suspicious. I'm going to try reinstalling the bootloader on the nonfunctional controllers

isidroas, 2023-05-08:
`#define SPLIT_USB_DETECT` in `config.h`

source: https://www.reddit.com/r/olkb/comments/oh65me/comment/h4nbqzo/?utm_source=share&utm_medium=web2x&context=3