omkbd/ErgoDash #27: Plans to add revision 1.3 or 2.0 PCB schematics and spec?

richerve, 2019-10-02:
I've seen in your twitter account @omkbd pictures of PCBs version 1.3 and 2.0, this last one with hot-swappable switches and what it looks like a usb-c interconnect between the two sides. Do you have any plan on releasing the source for those revisions or are just private iterations not ready to be shared?

richerve, 2019-10-05:
Ping @omkbd

JimmyMultani, 2019-10-20:
Hey @omkbd, any plans to release these schematics? Thank you.

MattBlissett, 2019-10-28:
I'd like to make an Ergodash.  Are the two changes above (hot-swappable switches, USB-C) and the rotary encoder (from the other issue and Twitter) the changes expected in r1.3 / r2.0, or is there more?

lucasfperotti, 2019-11-06:
@omkbd Oh my god yes !!! I was about to buy a corne kbd kit just because i could not find a ergo dash with hot swap mx switches but if a 2.0 is going to happen i will get that !!! I am a great fan of this keyboard !

MattBlissett, 2019-11-07:
I just finished making my hot-swap Ergodash (rev 1.2). I used [Mill-Max sockets](https://www.mouser.com/ProductDetail/mill-max/0305-2-15-80-47-80-10-0/), as in [this build](https://www.reddit.com/r/MechanicalKeyboards/comments/bvm3pe/my_first_build_ergodash_with_hotswappable/) ([more detail](https://www.reddit.com/r/MechanicalKeyboards/comments/bvm6zf/my_first_build_ergodash_with_hotswappable/)).

I'm glad I used the slightly longer 0305 sockets, it was less fiddly than I expected. However, the socket for the key above the Pro Micro touches one of the Pro Micro's capacitors, so it's about 0.7mm higher than it should be. 

I'm not sure what Falba Tech use, as they sell the Ergodash preassembled with Mill-Max sockets.

Geobert, 2019-11-07:
What I'm super hype on this 2.0 is the 4 keys thumb cluster, so I really hope 2.0 will become a thing!

richerve, 2019-11-12:
@omkbd any update on this?

omkbd, 2019-11-28:
I am very busy now. You will be able to resume development in a little more time.

n8gray, 2019-12-03:
Thanks for the update- count me in with the others who would be thrilled by a socketed design.

NellyWhadsDev, 2020-01-21:
Hey @omkbd any chance this is coming soon? If you need help, I'm more than happy to pitch in any way I can. Dying to get a v2.0 (at least from the teaser pics)

MMMikeM, 2020-07-03:
@omkbd is the project still active?

richerve, 2021-05-19:
As this was mainly a question and haven't been answered in more than a year, will close it.

closed 2021-05-19 by richerve