omkbd/ErgoDash #33: M2 spacers height

GioTB, 2020-09-13:
Hi!, does anyone know what is the height of the m2 spacers?

clembo1, 2020-09-14:
Direct translation from Japanese ala google
https://github.com/omkbd/ErgoDash/blob/master/Doc/Shoplist_JP.md
M2 x 5 mm screw x 22
M2 x 8mm screw x 6
M2 x 6mm spacer x 14

Math:  2 screws per spacer (1 bottom, 1 top)  six (6) 8MM screws uses 3 spacers.  This leaves 11 spacers left.  22 screws / 2 = 11 spacers.  Thus all screws are accounted for.

Only caveat is that the directions on the English guide, Step 11, references "Install the 5mm and 6mm spacers from the top."  Just my guess, but I think it means 5MM screws with the 6MM spacer.  

I would also add that the English Parts list for the Screws and Spacers added a lot of confusion for me till I thought things through "M2×8mm screw × 6 M2 spacer × 14".  I think it was typos or translation issue from google translate or some other service.

MattBlissett, 2020-09-14:
The pull request I made fixes all three things -- the missing "mm" after 6, the missing newline, and the missing word "screws" in the build guide.  @omkbd will probably merge it when he sees the notification.

GioTB, 2020-09-14:
Thanks!!!!, that helps a lot!

PR #34 (2020-09-18): Spacers: make the English list and guide the same as the Japanese.
Resolves #26
Resolves #33
commit 7b7924c1 (2020-09-18): Spacers: make the English list and guide the same as the Japanese.

Resolves #26
Resolves #33
closed 2020-09-18 by omkbd