omkbd/ErgoDash #42: Alternative for the MJTP1117 switch

MaKaNu, 2021-01-27:
I have some trouble to find the switches. There are some reseller like mouser or digikey wit a decend price, but the transport costs are absurdly high for part in the cents area.

I also never saw this kind of switches around  europe.

Any Ideas for alternative switches, which are as best available around the world?

Strayer, 2021-01-28:
I also had trouble finding these *anywhere* with reasonable shipping. In the end I ordered them on AliExpress. Sadly the ones I ordered are not available anymore (https://de.aliexpress.com/item/32964192707.html), but it should be possible to find them there. I can send you some leftovers if you are interested, I should still have some of them – email me at strayer@olle-orks.org if you are interested.

MaKaNu, 2021-01-28:
The AliExpress link helped a lot. For everyone else looking for the MJTP: "switch side press" will do the job. @Strayer thanks for offer.

closed 2021-01-28 by MaKaNu