omkbd/ErgoDash #44: Reset is not working while flashing

MaKaNu, 2021-06-16:
Hi, 

I builded Rev1.2 with a few parts switched (Different LEDs because Kailh switches, different SMD LEDs SK6812 because they seemed a good option for the ws2812b). I finally installed all parts and also got my self-made TRRS cable running (I Tested the connections and there is no error in the cable).
My qmk setup:

```bash
multibuild.keymap=default
user.keyboard=ergodash/rev1
user.keymap=makanu
```

And the keymap makanu is at the moment just a copy of the default keymap.

When I use the command ```qmk flash```, the command ends up with ```Detecting USB port, reset your controller now...``` and pressing the reset button doesn't do anything.

Additional I recognized that the switch LEDs on the right side are dimmed. This also happens if I switch master side. 

Any hint where I should start measuring with a multimeter, to provide further informations?

MaKaNu, 2022-09-27:
Sadly not until now. My first idea was faulty soldering, but I was unable to find the issue. At the moment I have a de-soldered micro controller which is probably defect, because of the de-soldering. Sadly the project is on stall for me at the moment. I am also not really happy about the soldering position of the micro-controller, since it is so close to the pins.

JonahBranch, 2022-09-27:
I'm wondering if it's an issue with the bootloader. I'm going to try rewriting the bootloader and I'll let you know if that fixes it