omkbd/ErgoDash #46: Underglow RGB LEDs not working on both sides

ryzhkovvv, 2021-07-22:
LEDS are off when I plug the keyboard and don't turn on on the RGB Toggle key press.

This is my first keyboard build, so it might be some stupid mistake, but I'm pretty sure my keyboard soldered correctly, LED jumpers are bridged according to manual and I just refreshed soldering joints on MOSFET, and LEDs. They seem fine.

Here is the photo of the left (master) half of the keyboard:
[image]

usb79, 2021-07-23:
Hi, 
guess I had the same or a similar issue. In my case it was the bus controlling the WS2812 LEDs. Check the connection between each output and input if the solder joints are ok. Measuring between output and corresponding next input of the WS2812 LEDs (on top of the connectors) helps verifying if the LEDs are connected properly.
Good luck, I hope you get it to work soon!

ryzhkovvv, 2021-07-23:
Solved! I'm an idiot, I was flashing my keyboard with firmware in which RGB underflow was disabled, so  RGB Toggle key didn't work either. Now I figured out how to properly turn my json file from Configurator to keymap.c with json2c command in QMK SYS and compiled a normal firmware my underflow works normally.
It was pretty unintuitive, though - that if RGB_Enable is set to 'no' underflow not just disabled by default, but whole RGB functionality is disabled.

closed 2021-07-24 by ryzhkovvv